CrazyIvan359/mqttany #21: Error on MCP23017 init

tobof, 2019-09-25:
I'm trying to get an MCP23017 connected to a RPi 3 to work.

I'm using this minimalistic config:
```yaml
mqtt:

  ### Broker
  host: 192.168.2.13 # address or hostname

######## MCP230xx Module ########
mcp230xx:

  device_section:

    bus id: '1'
    address: 0x20
    device: MCP23017
  
    single_pin:
      pin: 0
```

Verbose output: 
```
root@i2cRelay:/opt/mqttany# python3 mqttany/mqttany.py --verbose
2019-09-25 13:19:48,197 [INFO ] [mqttany ] [mqttany                 ] MQTTany 0.7.0 starting
2019-09-25 13:19:48,198 [DEBUG] [mqttany ] [mqttany                 ] Loading config
2019-09-25 13:19:48,207 [DEBUG] [mqttany ] [mqttany                 ] Loading module 'mqtt'
2019-09-25 13:19:48,321 [DEBUG] [mqttany ] [mqttany.mqtt            ] Parsing config
2019-09-25 13:19:48,321 [DEBUG] [mqttany ] [mqttany.mqtt            ] Got value '192.168.2.13' for config option 'host'
2019-09-25 13:19:48,322 [DEBUG] [mqttany ] [mqttany.mqtt            ] Using default value '1883' for config option 'port'
2019-09-25 13:19:48,322 [DEBUG] [mqttany ] [mqttany.mqtt            ] Using default value '{hostname}' for config option 'client id'
2019-09-25 13:19:48,323 [DEBUG] [mqttany ] [mqttany.mqtt            ] Using default value '' for config option 'username'
2019-09-25 13:19:48,323 [DEBUG] [mqttany ] [mqttany.mqtt            ] Using default value '' for config option 'password'
2019-09-25 13:19:48,324 [DEBUG] [mqttany ] [mqttany.mqtt            ] Using default value '0' for config option 'qos'
2019-09-25 13:19:48,324 [DEBUG] [mqttany ] [mqttany.mqtt            ] Using default value 'False' for config option 'retain'
2019-09-25 13:19:48,325 [DEBUG] [mqttany ] [mqttany.mqtt            ] Using default value '{client_id}' for config option 'root topic'
2019-09-25 13:19:48,325 [DEBUG] [mqttany ] [mqttany.mqtt            ] Using default value 'lwt' for config option 'lwt topic'
2019-09-25 13:19:48,326 [DEBUG] [mqttany ] [mqttany.mqtt            ] Config loaded
2019-09-25 13:19:48,326 [INFO ] [mqttany ] [mqttany                 ] Module 'mqtt' loaded successfully
2019-09-25 13:19:48,327 [DEBUG] [mqttany ] [mqttany                 ] Loading module 'mcp230xx'
2019-09-25 13:19:48,388 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Board is RASPBERRY_PI_3B_PLUS
2019-09-25 13:19:48,389 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Parsing config
2019-09-25 13:19:48,390 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Using default value '{module_name}' for config option 'topic'
2019-09-25 13:19:48,390 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Using default value 'set' for config option 'topic set'
2019-09-25 13:19:48,390 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Using default value 'poll' for config option 'topic get'
2019-09-25 13:19:48,391 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Using default value 'ON' for config option 'payload on'
2019-09-25 13:19:48,391 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Using default value 'OFF' for config option 'payload off'
2019-09-25 13:19:48,392 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Using default value '0.0' for config option 'polling interval'
2019-09-25 13:19:48,392 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Found regex '.+' in options
2019-09-25 13:19:48,393 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Config key 'device_section' matched to options regex '.+'
2019-09-25 13:19:48,394 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Got value '1' for config option 'bus id'
2019-09-25 13:19:48,394 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Got value '32' for config option 'address'
2019-09-25 13:19:48,395 [ERROR] [mqttany ] [mqttany.mcp230xx        ] Missing config option 'chip'
2019-09-25 13:19:48,395 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Found regex '.+' in options
2019-09-25 13:19:48,396 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Config key 'device' matched to options regex '.+'
2019-09-25 13:19:48,396 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Config key 'single_pin' matched to options regex '.+'
2019-09-25 13:19:48,397 [WARN ] [mqttany ] [mqttany.mcp230xx        ] Optional section 'device_section' is not valid, it will be ignored
2019-09-25 13:19:48,397 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Config loaded
2019-09-25 13:19:48,398 [INFO ] [mqttany ] [mqttany                 ] Module 'mcp230xx' loaded successfully
2019-09-25 13:19:48,398 [DEBUG] [mqttany ] [mqttany                 ] Creating process for 'mqtt'
2019-09-25 13:19:48,399 [DEBUG] [mqttany ] [mqttany                 ] Process created successfully for module 'mqtt'
2019-09-25 13:19:48,399 [DEBUG] [mqttany ] [mqttany                 ] Starting process for 'mqtt'
2019-09-25 13:19:48,404 [INFO ] [mqttany ] [mqttany                 ] Process started successfully for module 'mqtt'
2019-09-25 13:19:48,406 [DEBUG] [mqttany ] [mqttany                 ] Creating process for 'mcp230xx'
2019-09-25 13:19:48,407 [DEBUG] [mqttany ] [mqttany                 ] Process created successfully for module 'mcp230xx'
2019-09-25 13:19:48,407 [DEBUG] [mqttany ] [mqttany                 ] Starting process for 'mcp230xx'
2019-09-25 13:19:48,408 [DEBUG] [mqtt    ] [mqttany.mqtt            ] Creating MQTT client
2019-09-25 13:19:48,413 [DEBUG] [mqtt    ] [mqttany.mqtt            ] Attaching callbacks
2019-09-25 13:19:48,412 [INFO ] [mqttany ] [mqttany                 ] Process started successfully for module 'mcp230xx'
[... 17 lines cut ...]
2019-09-25 13:20:01,964 [DEBUG] [mqttany ] [mqttany                 ] Unloading module 'mcp230xx'
2019-09-25 13:20:01,966 [DEBUG] [mqttany ] [mqttany                 ] Stopping subprocess for 'mcp230xx' with 10s timeout
2019-09-25 13:20:01,996 [DEBUG] [mcp230xx] [mqttany.mcp230xx        ] Received poison pill
2019-09-25 13:20:02,003 [DEBUG] [mqttany ] [mqttany                 ] Subproccess for module 'mcp230xx' stopped cleanly
2019-09-25 13:20:02,004 [INFO ] [mqttany ] [mqttany                 ] Module 'mcp230xx' unloaded
2019-09-25 13:20:02,005 [DEBUG] [mqttany ] [mqttany                 ] Unloading module 'mqtt'
2019-09-25 13:20:02,006 [DEBUG] [mqttany ] [mqttany                 ] Stopping subprocess for 'mqtt' with 10s timeout
2019-09-25 13:20:02,020 [DEBUG] [mqtt    ] [mqttany.mqtt            ] Received poison pill
2019-09-25 13:20:02,021 [DEBUG] [mqtt    ] [mqttany.mqtt            ] Disconnecting
2019-09-25 13:20:02,022 [DEBUG] [mqtt    ] [mqttany.mqtt.client     ] Sending PUBLISH (d0, q0, r1, m3), 'b'i2cRelay/lwt'', ... (7 bytes)
2019-09-25 13:20:02,024 [DEBUG] [mqtt    ] [mqttany.mqtt.client     ] Sending DISCONNECT
2019-09-25 13:20:02,026 [DEBUG] [mqtt    ] [mqttany.mqtt            ] Stopping MQTT client loop
2019-09-25 13:20:02,034 [DEBUG] [mqttany ] [mqttany                 ] Subproccess for module 'mqtt' stopped cleanly
2019-09-25 13:20:02,035 [INFO ] [mqttany ] [mqttany                 ] Module 'mqtt' unloaded
2019-09-25 13:20:02,036 [INFO ] [mqttany ] [mqttany                 ] MQTTany ```

This warning looks suspicious:
```
2019-09-25 13:19:48,397 [WARN ] [mqttany ] [mqttany.mcp230xx        ] Optional section 'device_section' is not valid, it will be ignored
```
Looks like it is not entering [this loop](https://github.com/CrazyIvan359/mqttany/blob/a1992b2a3aa23ccb7f0d4502d8d0f6e551a0e35c/mqttany/modules/mcp230xx.py#L162) to initialize the device.

Additionally the example in the Wiki shows "chip" where "device" is meant?

Keep up the great work! Amazing!

CrazyIvan359, 2019-09-25:
Might have been easier to spot with verbose off
`Missing config option 'chip'`

I'll be creating a new `trace` ('-vv') log level soon to reduce the number of entries somewhat.

I see your confusion though, seems I forgot to update the example config file with the new name. I changed it because I'm using the 'device' key internally. 

Which do you think is more obvious, device or chip?

tobof, 2019-09-25:
Thanks a lot for your fast response!

I would prefer "chip" but I'm okay with "device" too. :smile: 

Just changing `device` to `chip` does not do the trick for me:

```
2019-09-25 13:53:03,719 [INFO ] [mqttany                 ] MQTTany 0.7.0 starting
2019-09-25 13:53:03,841 [INFO ] [mqttany                 ] Module 'mqtt' loaded successfully
2019-09-25 13:53:03,904 [ERROR] [mqttany                 ] An exception occurred while running function 'init'
2019-09-25 13:53:03,905 [ERROR] [mqttany                 ] Traceback (most recent call last):
2019-09-25 13:53:03,905 [ERROR] [mqttany                 ] 
2019-09-25 13:53:03,906 [ERROR] [mqttany                 ]   File "/opt/mqttany/mqttany/modules/__init__.py", line 120, in _call_func
2019-09-25 13:53:03,906 [ERROR] [mqttany                 ]     retval = func(**kwargs)
2019-09-25 13:53:03,907 [ERROR] [mqttany                 ] 
2019-09-25 13:53:03,907 [ERROR] [mqttany                 ]   File "/opt/mqttany/mqttany/modules/mcp230xx.py", line 167, in init
2019-09-25 13:53:03,908 [ERROR] [mqttany                 ]     bus_id=device_config[CONF_KEY_BUS_ID], options=device_config))
2019-09-25 13:53:03,908 [ERROR] [mqttany                 ] 
2019-09-25 13:53:03,908 [ERROR] [mqttany                 ] KeyError: 'name'
2019-09-25 13:53:03,909 [ERROR] [mqttany                 ] 
2019-09-25 13:53:03,909 [WARN ] [mqttany                 ] Module 'mcp230xx' not initialized
2019-09-25 13:53:03,914 [INFO ] [mqttany                 ] Process started successfully for module 'mqtt'
2019-09-25 13:53:03,938 [INFO ] [mqttany.mqtt            ] Connected to broker '192.168.2.13:1883'
```

CrazyIvan359, 2019-09-25:
Doh! Must have been a late night when I did that.

Add the following to line 167 of `modules/mcp230xx.py`
```python
device_name=device_name,
```

I'll push a fix tonight. I'm a bit hampered for testing anything other than GPIO stuff right now. My test Pi has burned out yet another SD card.

tobof, 2019-09-25:
Looks like this will solve the problem.
Thank you!

CrazyIvan359, 2019-09-25:
Fixed in v0.7.1

closed 2019-09-25 by CrazyIvan359