CrazyIvan359/mqttany #31: [MCP23017] mqttany stops switching pins

tobof, 2019-10-02:
**Describe the bug**
I've 16 relays connected to a MCP23017.
mqttany is working fine for a few times switching the pins ON and OFF as expected.
But after a few changes mqttany stops propagating the changes to the hardware.

The log (trace) looks fine and I see no difference between the initial state changes that work and the following state changes that does not work. 

So the problem is: mqttany with my MCP23017 works fine for a few times: 5-10 times (maybe) and then stops changing the pins on the MCP23017. After a restart of mqttany it continues to work again till it fails again. mqttany does not recover itself.

Changing the pins via i2set does not show this behaviour.

This is somewhat hard to describe. Do you have an idea how to investigate this further?

**Environment**
- MQTTany version: MQTTany 0.8.0
- Platform: Raspberry Pi 3B+*
- Modules:
  - [ ] GPIO
  - [x] MCP230xx
  - [ ] OneWire

**Logs**
```
root@i2cRelay:/opt/mqttany# python3 mqttany/mqttany.py -v
2019-10-02 13:52:34,139 [INFO ] [mqttany ] [mqttany                 ] MQTTany 0.8.0 starting
2019-10-02 13:52:34,140 [DEBUG] [mqttany ] [mqttany                 ] Loading config file
2019-10-02 13:52:34,156 [DEBUG] [mqttany ] [mqttany                 ] Loading module 'mqtt'
2019-10-02 13:52:34,269 [DEBUG] [mqttany ] [mqttany.mqtt            ] Parsing config
2019-10-02 13:52:34,270 [DEBUG] [mqttany ] [mqttany.mqtt            ] Config loaded
2019-10-02 13:52:34,271 [DEBUG] [mqttany ] [mqttany                 ] Module 'mqtt' loaded successfully
2019-10-02 13:52:34,271 [DEBUG] [mqttany ] [mqttany                 ] Loading module 'mcp230xx'
2019-10-02 13:52:34,333 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Board is RASPBERRY_PI_3B_PLUS
2019-10-02 13:52:34,334 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Parsing config
2019-10-02 13:52:34,335 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Config loaded
2019-10-02 13:52:34,336 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configuring MCP23017 'relayBoard1' at address 0x20 on I2C bus '1' with options: {'bus id': 1, 'address': 32, 'chip': 'MCP23017', 'topic': '{device_name}', 'single_pin': {'pin': [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15], 'topic': '{pin}', 'first index': 0, 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': '{payload_off}'}, 'name': 'relayBoard1'}
2019-10-02 13:52:34,337 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured GP00 on 'MCP23017' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/0', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-02 13:52:34,338 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured GP01 on 'MCP23017' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/1', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-02 13:52:34,338 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured GP02 on 'MCP23017' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/2', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-02 13:52:34,339 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured GP03 on 'MCP23017' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/3', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-02 13:52:34,340 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured GP04 on 'MCP23017' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/4', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-02 13:52:34,341 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured GP05 on 'MCP23017' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/5', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-02 13:52:34,342 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured GP06 on 'MCP23017' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/6', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-02 13:52:34,342 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured GP07 on 'MCP23017' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/7', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-02 13:52:34,343 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured GP08 on 'MCP23017' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/8', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-02 13:52:34,344 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured GP09 on 'MCP23017' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/9', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-02 13:52:34,344 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured GP10 on 'MCP23017' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/10', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-02 13:52:34,345 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured GP11 on 'MCP23017' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/11', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-02 13:52:34,346 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured GP12 on 'MCP23017' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/12', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-02 13:52:34,347 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured GP13 on 'MCP23017' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/13', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-02 13:52:34,347 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured GP14 on 'MCP23017' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/14', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-02 13:52:34,348 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured GP15 on 'MCP23017' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/15', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-02 13:52:34,349 [DEBUG] [mqttany ] [mqttany                 ] Module 'mcp230xx' loaded successfully
2019-10-02 13:52:34,354 [INFO ] [mqttany ] [mqttany                 ] Process started successfully for module 'mqtt'
2019-10-02 13:52:34,357 [DEBUG] [mqtt    ] [mqttany.mqtt            ] Creating MQTT client
2019-10-02 13:52:34,359 [INFO ] [mqttany ] [mqttany                 ] Process started successfully for module 'mcp230xx'
2019-10-02 13:52:34,363 [DEBUG] [mqtt    ] [mqttany.mqtt            ] Attaching callbacks
2019-10-02 13:52:34,364 [DEBUG] [mcp230xx] [mqttany.mcp230xx        ] Setting up hardware
2019-10-02 13:52:34,364 [DEBUG] [mqtt    ] [mqttany.mqtt            ] Queuing connect event
2019-10-02 13:52:34,365 [DEBUG] [mqtt    ] [mqttany.mqtt            ] Starting MQTT client thread
2019-10-02 13:52:34,365 [DEBUG] [mcp230xx] [mqttany.mcp230xx        ] Initializing I2C bus '1'
2019-10-02 13:52:34,377 [DEBUG] [mqtt    ] [mqttany.mqtt.client     ] Sending CONNECT (u1, p1, wr1, wq0, wf1, c0, k15) client_id=b'i2cRelay'
2019-10-02 13:52:34,380 [DEBUG] [mqtt    ] [mqttany.mqtt.client     ] Received CONNACK (1, 0)
2019-10-02 13:52:34,381 [INFO ] [mqtt    ] [mqttany.mqtt            ] Connected to broker '192.168.2.13:1883'
2019-10-02 13:52:34,381 [DEBUG] [mcp230xx] [mqttany.mcp230xx        ] Successfully initialized I2C bus '1'
[... cut ...]
2019-10-02 13:57:52,120 [DEBUG] [mqtt    ] [mqttany.mqtt.client     ] Sending PUBLISH (d0, q0, r1, m100), 'b'i2cRelay/lwt'', ... (7 bytes)
2019-10-02 13:57:52,122 [DEBUG] [mqtt    ] [mqttany.mqtt.client     ] Sending DISCONNECT
2019-10-02 13:57:52,125 [DEBUG] [mqtt    ] [mqttany.mqtt            ] Stopping MQTT client loop
2019-10-02 13:57:52,132 [DEBUG] [mqttany ] [mqttany                 ] Subproccess for module 'mqtt' stopped cleanly
2019-10-02 13:57:52,133 [INFO ] [mqttany ] [mqttany                 ] Module 'mqtt' unloaded
2019-10-02 13:57:52,134 [INFO ] [mqttany ] [mqttany                 ] MQTTany stopped
```

CrazyIvan359, 2019-10-02:
If you have the time, setup a test config file (you can specify a config file as an arg, see `mqttany.py -h`) with a single pin configured and repeat this test and note exactly how many cycles work before it stops.

This will be a hard one to track, as you say. My initial suspicion is the Adafruit library I'm using. Any chance you have an I2C sniffer? 

I've been considering options for adding additional I2C device support and might remove the MCP23xxx module in favour of an I2C module that supports devices. See the new OneWire module for an idea of the structure. Such a module would have a generic I2C interface and device submodules that would use that interface to communicate with devices.

CrazyIvan359, 2019-10-05:
Hey @tobof have you had a change to test this more?

If I write a small program to cycle a pin and record the number of times it's been cycled would you be willing to do some tests with it?

tobof, 2019-10-05:
I've already written a little script to cycle the pins and the results are:

On GPB there are 8 relays attached but these aren't connected yet. Switching those relays works just fine. 

On GPA I've 8 relays connected which control four shutters. They get stuck after a random number of pin cycles. Something between 5 and 15 times and after a while without any pin change.
The relays are connected through optocoupler with complete separate power rails. 

Based on your idea of a problem in the adafruit library I've written a script to control my shutters based on that library. That script is working flawlessly. No problem at all. Because the script is called for every action (shutter up, down, stop). It is initialized at every start. 
My guess is: something happens between initialization of the pin and the change of the pin that isn't working if I use mqttany. 

I have no i2c sniffer and if I build one I'm not that deep into i2c that I expect to see the problem. ;-)

CrazyIvan359, 2019-10-05:
Interesting. Put an LED on a pin on port B and modify your program to cycle that pin indefinitely with a 5 second delay or so, printing a count of how many times the pin has been cycled. That way the library will stay loaded.

tobof, 2019-10-05:
I've cycled the pins on GPB about 100 times without any problem with mqqtany. Beside the relay at every pin a LED is attached. So this looks like a problem with GPA or my setup.
I need to write a script based on the Adafruit library that switches GPA and keeps loaded/initialized.
I have tried to build a separate environment to exclude the relays but I only have a RPi 1b and RPi 4 which aren't working with mqttany.

CrazyIvan359, 2019-10-05:
I might be able to help modify the script if you post it here.

The 1B should work. The 4 would require Platform Detect to be able to identify it, which I don't recall if it can. If it can't then I should be able to add it and submit a PR. I will look into it.

tobof, 2019-10-14:
After some testing I've no clue what's going on! :laughing: 

So here are the results:

Two scripts doing both:

Move the shutter for 1 second down, pause 1 second and move it down again for 10 times. After that the shutter is moved up for 1 second and after a pause of 1 second moved up again for 10 times.

**1. script** running with mqttany currently in Version 0.8.2 but the Version doesn't seem to matter:
```
#!/bin/bash

echo "DOWN!"

mosquitto_pub -h 192.168.2.13 -p 1883 -m "OFF" -t "i2cRelay/mcp230xx/relayBoard1/0/set"
for (( i=0 ; i<10 ; i++ ))
do
   echo "Loop cycle: $i"
  	mosquitto_pub -h 192.168.2.13 -p 1883 -m "ON" -t "i2cRelay/mcp230xx/relayBoard1/1/set"
   sleep 1s
   mosquitto_pub -h 192.168.2.13 -p 1883 -m "OFF" -t "i2cRelay/mcp230xx/relayBoard1/1/set"
   sleep 1s
done

echo "UP!"

mosquitto_pub -h 192.168.2.13 -p 1883 -m "OFF" -t "i2cRelay/mcp230xx/relayBoard1/1/set"
for (( i=0 ; i<10 ; i++ ))
do
   echo "Loop cycle: $i"
  	mosquitto_pub -h 192.168.2.13 -p 1883 -m "ON" -t "i2cRelay/mcp230xx/relayBoard1/0/set"
   sleep 1s
   mosquitto_pub -h 192.168.2.13 -p 1883 -m "OFF" -t "i2cRelay/mcp230xx/relayBoard1/0/set"
   sleep 1s
done
```

**2. script** without mqttany but using the adafruit library:
```
import time

import board
import busio
import digitalio
import sys

from adafruit_mcp230xx.mcp23017 import MCP23017

i2c = busio.I2C(board.SCL, board.SDA)

mcp = MCP23017(i2c, address=0x20)

pin0 = mcp.get_pin(0)
pin1 = mcp.get_pin(1)

pin0.switch_to_output(value=True)
pin1.switch_to_output(value=True)

print("DOWN!")
pin0.value = True
for x in range(0, 10):
    print("Loop cycle: %d" % x)
    pin1.value = False
    time.sleep(1.0)
    pin1.value = True
    time.sleep(1.0)

pin1.value = True
print("UP!")
for x in range(0, 10):
    print("Loop cycle: %d" % x)
    pin0.value = False
    time.sleep(1.0)
    pin0.value = True
    time.sleep(1.0)
```

I've connected four shutters via relais with optocoupler to A0 to A7. If the shutters are "connected" mqttany gets stuck after 2 to 8 pin changes. There's no output in the log file that is helping here. If the shutters aren't connected (main switch to off) the cycle works fine till the end. No hiccup!

The second script (only adafruit library) works fine every time. Absolutly no hiccup. 

So a shot in the dark:

With the shutters connected via relais (although there are optocoupler in between and MCP23017 with RPi does not share any power line with the relais, there are separate dc lines for both) there is some recoil to the i2c connection that leads to a failure in mqttany. But why is my simple script working fine?

CrazyIvan359, 2019-10-14:
Thank you doing these thorough tests! I come to the same conclusion as you though, doesn't make any sense...

And there are no log entires from MQTTany saying it couldn't read from the device? Does it still show log entries that it has written to the device after it stops responding? If not then maybe the MCP23xxx thread hit a segfault and died silently. When you exit does it say the MCP23xxx process was terminated cleanly or that it wasnt running?

Also for your reference I will be going ahead with the general I2C module that will be wrapping the smbus2 library and writing my own device modules. Don't take that to mean I'm not going to help find the problem here, because if it's with MQTTany then it is likely to affect the new I2C module as well.

tobof, 2019-10-14:
Here is the log output. It stopped after the first cycle. First cycle worked fine. Second and third cycle didn't work.

```
root@i2cRelay:/opt/mqttany# python3 mqttany/mqttany.py -v
2019-10-14 13:52:54,704 [INFO ] [mqttany ] [mqttany                 ] MQTTany 0.8.2 starting
2019-10-14 13:52:54,705 [DEBUG] [mqttany ] [mqttany                 ] Loading config file
2019-10-14 13:52:54,721 [DEBUG] [mqttany ] [mqttany                 ] Loading module 'mqtt'
2019-10-14 13:52:54,834 [DEBUG] [mqttany ] [mqttany.mqtt            ] Parsing config
2019-10-14 13:52:54,835 [DEBUG] [mqttany ] [mqttany.mqtt            ] Config loaded
2019-10-14 13:52:54,836 [DEBUG] [mqttany ] [mqttany                 ] Module 'mqtt' loaded successfully
2019-10-14 13:52:54,836 [DEBUG] [mqttany ] [mqttany                 ] Loading module 'mcp230xx'
2019-10-14 13:52:54,903 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Board is RASPBERRY_PI_3B_PLUS
2019-10-14 13:52:54,903 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Parsing config
2019-10-14 13:52:54,905 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Config loaded
2019-10-14 13:52:54,906 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configuring MCP23017 'relayBoard1' at address 0x20 on I2C bus '1' with options: {'bus id': 1, 'address': 32, 'chip': 'MCP23017', 'topic': '{device_name}', 'single_pin': {'pin': [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15], 'topic': '{pin}', 'first index': 0, 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': '{payload_off}'}, 'name': 'relayBoard1'}
2019-10-14 13:52:54,906 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured 'single_pin' on GP00 on 'MCP23017' 'relayBoard1' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/0', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-14 13:52:54,907 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured 'single_pin' on GP01 on 'MCP23017' 'relayBoard1' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/1', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-14 13:52:54,908 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured 'single_pin' on GP02 on 'MCP23017' 'relayBoard1' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/2', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-14 13:52:54,909 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured 'single_pin' on GP03 on 'MCP23017' 'relayBoard1' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/3', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-14 13:52:54,909 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured 'single_pin' on GP04 on 'MCP23017' 'relayBoard1' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/4', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-14 13:52:54,910 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured 'single_pin' on GP05 on 'MCP23017' 'relayBoard1' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/5', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-14 13:52:54,911 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured 'single_pin' on GP06 on 'MCP23017' 'relayBoard1' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/6', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-14 13:52:54,911 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured 'single_pin' on GP07 on 'MCP23017' 'relayBoard1' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/7', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-14 13:52:54,912 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured 'single_pin' on GP08 on 'MCP23017' 'relayBoard1' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/8', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-14 13:52:54,913 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured 'single_pin' on GP09 on 'MCP23017' 'relayBoard1' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/9', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-14 13:52:54,913 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured 'single_pin' on GP10 on 'MCP23017' 'relayBoard1' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/10', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
2019-10-14 13:52:54,914 [DEBUG] [mqttany ] [mqttany.mcp230xx        ] Configured 'single_pin' on GP11 on 'MCP23017' 'relayBoard1' at address 0x20 with options: {'name': 'single_pin', 'topic': 'i2cRelay/mcp230xx/relayBoard1/11', 'direction': digitalio.Direction.OUTPUT, 'resistor': 'off', 'invert': True, 'initial state': 'OFF'}
[... cut ...]
2019-10-14 13:51:37,011 [INFO ] [mqttany                 ] Process started successfully for module 'mcp230xx'
2019-10-14 13:51:37,027 [INFO ] [mqttany.mqtt            ] Connected to broker '192.168.2.13:1883'
2019-10-14 13:51:37,033 [INFO ] [mqttany.mcp230xx        ] Initializing 'relayBoard1'
^C
2019-10-14 13:52:19,304 [INFO ] [mqttany                 ] Module 'mcp230xx' unloaded
2019-10-14 13:52:19,323 [INFO ] [mqttany                 ] Module 'mqtt' unloaded
2019-10-14 13:52:19,324 [INFO ] [mqttany                 ] MQTTany stopped
```

CrazyIvan359, 2019-10-14:
`Subproccess for module 'mcp230xx' stopped cleanly`

That line tells me that the process didn't die. Try running your 10 cycle test again and stopping MQTTany and see if you get that entry when it's exiting.

CrazyIvan359, 2019-11-28:
@tobof I have just finished a new I2C module, as I hinted that I was considering earlier. It currently only supports the MCP230xx devices and I have tested it with an MCP23017 I had on hand.

I have not stress tested it yet for repeatability, but my initial function tests worked great. If you are still interested in using MQTTany I would appreciate it if you would test it out. Just checkout the current commit on the `dev` branch, I just merged it.

There is no documentation available publicly yet, as it is still on the dev branch. You will need to move your current MCP230xx configuration into the `i2c` tree like below. The pin configurations remain the same, but you will need to change at least the key for `bus_id` to `bus`, and change the `chip` key to `device`.

```yaml
i2c:
  polling interval: 10
  mcp_test:
    device: 'mcp23017'
    bus: 0
    address: '0x20'
    mcp230xx:
      a3:
        pin: 3
        direction: 'output'
```

CrazyIvan359, 2019-11-29:
I wrote a test script to cycle a pin 500 times via MQTT and it worked perfectly.

Please let me know if you test this and have an issues with it.

CrazyIvan359, 2020-01-04:
This seems to be resolved when using the new I2C module that has replaced the MCP230xx module.

closed 2020-01-04 by CrazyIvan359