CrazyIvan359/mqttany #41: Bug Report

mountbaldy, 2020-01-23:
### Describe the bug
Raspberry PI Zero W is unsupported with the GPIO module.

### Expected behavior
Raspberry PI Zero W shares the 40 pin pin-out, as does every model after Rpi B+.  The Raspberry Pi Zero's do not come with soldered header pins, they are an add-on.

### Environment
- MQTTany version: 0.10.1
- Platform: Raspberry Pi Zero W

### Configuration
```yaml
# gpio:
  polling interval: 60
  debounce: 10

  channel1:
    pin: 18
    topic: '/{root_topic}/relay/1'
    direction: 'output'
    invert: true

  channel2:
    pin: 23
    topic: '/{root_topic}/relay/2'
    direction: 'output'
    invert: true

  channel3:
    pin: 24
    topic: '/{root_topic}/relay/3'
    direction: 'output'
    invert: true

  channel4:
    pin: 25
    topic: '/{root_topic}/relay/4'
    direction: 'output'
```

### Logs
```text

2020-01-23 10:25:55,707 [INFO ] [mqttany                 ] MQTTany 0.10.1 starting
2020-01-23 10:25:57,605 [ERROR] [mqttany.gpio            ] An error occurred while trying to import the Raspberry Pi GPIO library
2020-01-23 10:25:57,632 [ERROR] [mqttany.gpio            ] Traceback (most recent call last):
2020-01-23 10:25:57,637 [ERROR] [mqttany.gpio            ] 
2020-01-23 10:25:57,645 [ERROR] [mqttany.gpio            ]   File &quot;/opt/mqttany/mqttany/modules/gpio/GPIO/__init__.py&quot;, line 59, in getGPIO
2020-01-23 10:25:57,649 [ERROR] [mqttany.gpio            ]     from modules.gpio.GPIO.rpi import rpiGPIO
2020-01-23 10:25:57,656 [ERROR] [mqttany.gpio            ] 
2020-01-23 10:25:57,660 [ERROR] [mqttany.gpio            ]   File &quot;/opt/mqttany/mqttany/modules/gpio/GPIO/rpi.py&quot;, line 125, in &lt;module&gt;
2020-01-23 10:25:57,669 [ERROR] [mqttany.gpio            ]     adafruit_platformdetect.board.RASPBERRY_PI_A,
2020-01-23 10:25:57,674 [ERROR] [mqttany.gpio            ] 
2020-01-23 10:25:57,678 [ERROR] [mqttany.gpio            ] AttributeError: module &apos;adafruit_platformdetect.board&apos; has no attribute &apos;RASPBERRY_PI_A&apos;
2020-01-23 10:25:57,682 [ERROR] [mqttany.gpio            ] 
2020-01-23 10:25:57,699 [ERROR] [mqttany.gpio            ] Unsupported board RASPBERRY_PI_ZERO_W
2020-01-23 10:25:57,720 [INFO ] [mqttany                 ] Process started successfully for module &apos;mqtt&apos;
```

CrazyIvan359, 2020-01-25:
I've moved the request for support of the Zero to a different issue (#42) and I will use this one as a report of the missing attribute which is why the errors are occurring.

Did you also install the GPIO module and RPi interface requirements per their respective pages in the docs? If you did, run `pip3 show Adafruit-PlatformDetect` and show me the results.

mountbaldy, 2020-01-27:
Yes I did install the GPIO module and the RPi interface requirements.  I apologize my project was at the office. I finally grabbed it here this afternoon. 

Here are are the results for the above command:

`[17:37:46] openhabian@openhab:~$ pip3 show Adafruit-PlatformDetect
 
 Name: Adafruit-PlatformDetect
  Version: 2.1.0
  Summary: Platform detection for use by libraries like Adafruit-Blinka.
  Home-page: https://github.com/adafruit/Adafruit_Python_PlatformDetect
  Author: Adafruit Industries
  Author-email: circuitpython@adafruit.com
  License: MIT
  Location: /home/openhabian/.local/lib/python3.7/site-packages
  Requires: 
  Required-by: `

Thanks!  Let me know if you need me to do anything else.

CrazyIvan359, 2020-01-27:
Bingo! 5 days ago they updated to 2.x (GPIO module requires an older 1.x version) and the board constants have moved.

I will get my modules updated to use the 2.x version of Platform Detect as soon as I can. In the mean time you can run `pip3 install "Adafruit-PlatformDetect>=1.3.4,<2.0"` to downgrade your version of Platform Detect to the last 1.x version.

mountbaldy, 2020-01-27:
Awesome! I got things working by downgrading Adafruit-PlatformDetect.  Thank you for the help!  I'll look out for your update.

CrazyIvan359, 2020-02-08:
Fixed in [v0.10.2](https://github.com/CrazyIvan359/mqttany/releases/tag/v0.10.2)

closed 2020-02-08 by CrazyIvan359