CrazyIvan359/mqttany #79: GPIO Board Addition Request

adisD, 2020-10-30:
- Board: Raspberry PI 4

CrazyIvan359, 2020-10-31:
@adisD the issue template I linked to contains 3 commands that I require the output of. I need this information in order to add the board. 

From the template:
````
Output from `gpio readall` (if available):
```text

```

Output from `gpiodetect` (install `libgpiod`):
```text

```

Output from `gpioinfo` (install `libgpiod`):
```text

```
````

Veldhuis123, 2021-01-02:
Hello here the information requested,

Output from `gpio readall` (if available):
``
it's not available

``

Output from `gpiodetect` (install` libgpiod`):
``
gpiochip0 [pinctrl-bcm2835] (54 lines)
gpiochip1 [raspberrypi-exp-gpio] (8 lines)
gpiochip2 [ftdi-cbus] (4 lines)

``

Output from `gpioinfo` (install` libgpiod`):
``
gpiochip0 - 54 lines:
        line   0:     "ID_SDA"       unused   input  active-high 
        line   1:     "ID_SCL"       unused   input  active-high 
        line   2:       "SDA1"       unused   input  active-high 
        line   3:       "SCL1"       unused   input  active-high 
        line   4:  "GPIO_GCLK"       unused   input  active-high 
        line   5:      "GPIO5"       unused   input  active-high 
        line   6:      "GPIO6"       unused   input  active-high 
        line   7:  "SPI_CE1_N"       unused   input  active-high 
        line   8:  "SPI_CE0_N"       unused   input  active-high 
        line   9:   "SPI_MISO"       unused   input  active-high 
        line  10:   "SPI_MOSI"       unused   input  active-high 
        line  11:   "SPI_SCLK"       unused   input  active-high 
        line  12:     "GPIO12"       unused   input  active-high 
        line  13:     "GPIO13"       unused   input  active-high 
        line  14:       "TXD1"       unused   input  active-high 
        line  15:       "RXD1"       unused   input  active-high 
        line  16:     "GPIO16"       unused   input  active-high 
        line  17:     "GPIO17"       unused   input  active-high 
        line  18:     "GPIO18"       unused   input  active-high 
        line  19:     "GPIO19"       unused   input  active-high 
        line  20:     "GPIO20"       unused   input  active-high 
        line  21:     "GPIO21"       unused   input  active-high 
        line  22:     "GPIO22"       unused   input  active-high 
        line  23:     "GPIO23"       unused   input  active-high 
        line  24:     "GPIO24"       unused   input  active-high 
        line  25:     "GPIO25"       unused   input  active-high 
        line  26:     "GPIO26"       unused   input  active-high 
        line  27:     "GPIO27"       unused   input  active-high 
        line  28: "RGMII_MDIO"       unused   input  active-high 
        line  29:  "RGMIO_MDC"       unused   input  active-high 
        line  30:       "CTS0"       unused   input  active-high 
        line  31:       "RTS0"       unused   input  active-high 
        line  32:       "TXD0"       unused   input  active-high 
        line  33:       "RXD0"       unused   input  active-high 
        line  34:    "SD1_CLK"       unused   input  active-high 
        line  35:    "SD1_CMD"       unused   input  active-high 
        line  36:  "SD1_DATA0"       unused   input  active-high 
        line  37:  "SD1_DATA1"       unused   input  active-high 
        line  38:  "SD1_DATA2"       unused   input  active-high 
        line  39:  "SD1_DATA3"       unused   input  active-high 
        line  40:  "PWM0_MISO"       unused   input  active-high 
        line  41:  "PWM1_MOSI"       unused   input  active-high 
        line  42: "STATUS_LED_G_CLK" "led0" output active-high [used]
        line  43: "SPIFLASH_CE_N" unused input active-high 
        line  44:       "SDA0"       unused   input  active-high 
        line  45:       "SCL0"       unused   input  active-high 
        line  46: "RGMII_RXCLK" unused input active-high 
        line  47: "RGMII_RXCTL" unused input active-high 
        line  48: "RGMII_RXD0"       unused   input  active-high 
        line  49: "RGMII_RXD1"       unused   input  active-high 
        line  50: "RGMII_RXD2"       unused   input  active-high 
        line  51: "RGMII_RXD3"       unused   input  active-high 
        line  52: "RGMII_TXCLK" unused input active-high 
        line  53: "RGMII_TXCTL" unused input active-high 
gpiochip1 - 8 lines:
        line   0:      "BT_ON"       unused  output  active-high 
        line   1:      "WL_ON"       unused  output  active-high 
        line   2: "PWR_LED_OFF" "led1" output active-low [used]
        line   3: "GLOBAL_RESET" unused output active-high 
        line   4: "VDD_SD_IO_SEL" "vdd-sd-io" output active-high [used]
        line   5:   "CAM_GPIO"       unused  output  active-high 
        line   6:  "SD_PWR_ON" "sd_vcc_reg"  output  active-high [used]
        line   7:    "SD_OC_N"       unused   input  active-high 
gpiochip2 - 4 lines:
        line   0:      unnamed       unused   input  active-high 
        line   1:      unnamed       unused   input  active-high 
        line   2:      unnamed       unused   input  active-high 
        line   3:      unnamed       unused   input  active-high 

``

CrazyIvan359, 2021-01-07:
Please try [this branch](https://github.com/CrazyIvan359/mqttany/tree/add-rpi4) and let me know if that is working on your RPi4B

Veldhuis123, 2021-01-08:
Hello,

I am still learning with github, How can I scale to the new branch?

CrazyIvan359, 2021-01-08:
From the directory you downloaded MQTTany to (instructions list `/opt/mqttany`), run the following commands:

```bash
git fetch
git checkout add-rpi4
```

You can also use these commands to update to a new version as well, just replace `add-rpi4` with the name of the tag or branch you want to change to.

Veldhuis123, 2021-01-09:
Hello

i have set pin 26 as on output but he says as a input

```
pi@raspberrypi:/opt/mqttany $ python3 mqttany/mqttany.py
2021-01-09 16:17:48,537 [INFO ] [core                    ] MQTTany 0.13.0-dev starting
2021-01-09 16:17:48,537 [WARN ] [core                    ] 
2021-01-09 16:17:48,538 [WARN ] [core                    ] ########  DEVELOPMENT VERSION  ########
2021-01-09 16:17:48,538 [WARN ] [core                    ] #                                     #
2021-01-09 16:17:48,538 [WARN ] [core                    ] #  This is a development version and  #
2021-01-09 16:17:48,539 [WARN ] [core                    ] #    may be incomplete or unstable    #
2021-01-09 16:17:48,539 [WARN ] [core                    ] #                                     #
2021-01-09 16:17:48,539 [WARN ] [core                    ] #######################################
2021-01-09 16:17:48,540 [WARN ] [core                    ] 
2021-01-09 16:17:48,582 [INFO ] [core.gpio               ] Detected board: RASPBERRY_PI_4B
2021-01-09 16:17:48,761 [INFO ] [core                    ] Module 'mqtt' started successfully
2021-01-09 16:17:48,768 [INFO ] [core                    ] Module 'gpio' started successfully
2021-01-09 16:17:48,774 [INFO ] [gpio.digital            ] Setting up 'channel1' on GPIO26 as INPUT
2021-01-09 16:17:48,779 [INFO ] [mqtt                    ] Connected to broker '10.0.10.200:1883'

```

CrazyIvan359, 2021-01-09:
Please include the GPIO section of your config file

Veldhuis123, 2021-01-09:
this is my configuration,

```
version: '1.0'

mqtt:

  host: 10.0.10.200

gpio:
  polling interval: 60
  debounce: 10

  channel1:
    pin: 26
    topic: '/{root_topic}/relay/1'
    direction: 'output'
```

This is my openHab configuration
[image]

[image]

CrazyIvan359, 2021-01-09:
The configuration option is `pin mode` not `direction`

Veldhuis123, 2021-01-09:
Ten first thanks for all the help you offer me.

I changed it to pin mode and now it indicates this:
```
2021-01-09 17: 49: 46.112 [ERROR] [core.gpio.pin.digital] Unable to set GPIO26 with cdev, pin is in use
2021-01-09 17: 49: 46.113 [ERROR] [core.gpio.pin.digital] Cannot set GPIO26 with sysfs, pin is in use
```

CrazyIvan359, 2021-01-09:
Could you run again with the `-vv` flag. That will generate a long log, please post the entire log in your comment like this:

````
<details><summary>Trace log</summary>

```
PUT YOUR LOG HERE
```

</details>
````

Veldhuis123, 2021-01-09:
<details><summary>Trace log</summary>

```
2021-01-09 18:35:27,035 [INFO ] [core    ] [core                    ] MQTTany 0.13.0-dev starting
2021-01-09 18:35:27,036 [WARN ] [core    ] [core                    ] 
2021-01-09 18:35:27,036 [WARN ] [core    ] [core                    ] ########  DEVELOPMENT VERSION  ########
2021-01-09 18:35:27,037 [WARN ] [core    ] [core                    ] #                                     #
2021-01-09 18:35:27,037 [WARN ] [core    ] [core                    ] #  This is a development version and  #
2021-01-09 18:35:27,037 [WARN ] [core    ] [core                    ] #    may be incomplete or unstable    #
2021-01-09 18:35:27,038 [WARN ] [core    ] [core                    ] #                                     #
2021-01-09 18:35:27,038 [WARN ] [core    ] [core                    ] #######################################
2021-01-09 18:35:27,039 [WARN ] [core    ] [core                    ] 
2021-01-09 18:35:27,081 [INFO ] [core    ] [core.gpio               ] Detected board: RASPBERRY_PI_4B
2021-01-09 18:35:27,082 [DEBUG] [core    ] [core.gpio               ] Detected GPIO character device
2021-01-09 18:35:27,082 [DEBUG] [core    ] [config                  ] Loading config file
2021-01-09 18:35:27,116 [DEBUG] [core    ] [core                    ] Loading module 'mqtt'
2021-01-09 18:35:27,217 [DEBUG] [core    ] [core                    ] Module 'mqtt' is a communication module
2021-01-09 18:35:27,219 [DEBUG] [core    ] [mqtt                    ] Parsing config
2021-01-09 18:35:27,219 [TRACE] [core    ] [mqtt                    ] Got value '10.0.10.200' for config option 'host'
2021-01-09 18:35:27,220 [TRACE] [core    ] [mqtt                    ] Using default value '1883' for config option 'port'
2021-01-09 18:35:27,221 [TRACE] [core    ] [mqtt                    ] Using default value '{hostname}' for config option 'client id'
2021-01-09 18:35:27,222 [TRACE] [core    ] [mqtt                    ] Using default value '' for config option 'username'
2021-01-09 18:35:27,222 [TRACE] [core    ] [mqtt                    ] Using default value '' for config option 'password'
2021-01-09 18:35:27,223 [TRACE] [core    ] [mqtt                    ] Using default value '0' for config option 'qos'
2021-01-09 18:35:27,224 [TRACE] [core    ] [mqtt                    ] Using default value 'False' for config option 'retain'
2021-01-09 18:35:27,224 [TRACE] [core    ] [mqtt                    ] Using default value '{client_id}' for config option 'root topic'
2021-01-09 18:35:27,225 [TRACE] [core    ] [mqtt                    ] Using default value 'lwt' for config option 'lwt topic'
2021-01-09 18:35:27,226 [TRACE] [core    ] [mqtt                    ] Using default value '300' for config option 'heartbeat interval'
2021-01-09 18:35:27,226 [DEBUG] [core    ] [mqtt                    ] Config loaded successfully
2021-01-09 18:35:27,227 [DEBUG] [core    ] [core                    ] Module 'mqtt' loaded successfully
2021-01-09 18:35:27,228 [DEBUG] [core    ] [bus                     ] Module 'mqtt' added as a receiver
2021-01-09 18:35:27,230 [DEBUG] [core    ] [bus                     ] Module 'mqtt' added as a transmitter
2021-01-09 18:35:27,231 [DEBUG] [core    ] [core                    ] Loading module 'gpio'
2021-01-09 18:35:27,257 [DEBUG] [core    ] [core                    ] Module 'gpio' is an interface module
2021-01-09 18:35:27,259 [DEBUG] [core    ] [gpio                    ] Parsing config
2021-01-09 18:35:27,260 [TRACE] [core    ] [gpio                    ] Using default value 'Mode.SOC' for config option 'mode'
2021-01-09 18:35:27,260 [TRACE] [core    ] [gpio                    ] Got value '60' for config option 'polling interval'
2021-01-09 18:35:27,261 [TRACE] [core    ] [gpio                    ] Got value '10' for config option 'debounce'
2021-01-09 18:35:27,262 [TRACE] [core    ] [gpio                    ] Found regex '.+' in options
2021-01-09 18:35:27,263 [TRACE] [core    ] [gpio                    ] Config key 'channel1' matched to options regex '.+'
2021-01-09 18:35:27,264 [TRACE] [core    ] [gpio                    ] Descending into section 'channel1'
2021-01-09 18:35:27,265 [TRACE] [core    ] [gpio                    ] Got value '26' for config option 'pin'
2021-01-09 18:35:27,265 [TRACE] [core    ] [gpio                    ] Using default value '{pin_id}' for config option 'name'
2021-01-09 18:35:27,266 [TRACE] [core    ] [gpio                    ] Using default value '0' for config option 'first index'
2021-01-09 18:35:27,267 [TRACE] [core    ] [gpio                    ] Got selection 'output' for config option 'pin mode'
2021-01-09 18:35:27,267 [TRACE] [core    ] [gpio                    ] Using default value 'PinBias.NONE' for config option 'resistor'
2021-01-09 18:35:27,268 [TRACE] [core    ] [gpio                    ] Condition 'pin mode==PinMode.OUTPUT' matched for 'digital'
2021-01-09 18:35:27,269 [TRACE] [core    ] [gpio                    ] Descending into section 'digital'
2021-01-09 18:35:27,269 [TRACE] [core    ] [gpio                    ] Using default value 'PinEdge.NONE' for config option 'interrupt'
2021-01-09 18:35:27,270 [TRACE] [core    ] [gpio                    ] Using default value 'False' for config option 'invert'
2021-01-09 18:35:27,271 [TRACE] [core    ] [gpio                    ] Using default value 'False' for config option 'initial state'
[... cut ...]
2021-01-09 18:37:29,683 [TRACE] [mqtt    ] [mqtt.client             ] Sending SUBSCRIBE (d0, m185) [(b'raspberrypi/+/+/+/#', 0)]
2021-01-09 18:37:29,684 [DEBUG] [mqtt    ] [mqtt                    ] Heartbeat
2021-01-09 18:37:29,684 [TRACE] [mqtt    ] [mqtt.client             ] Sending PUBLISH (d0, q0, r1, m186), 'b'raspberrypi/lwt'', ... (6 bytes)
2021-01-09 18:37:29,685 [TRACE] [mqtt    ] [mqtt.client             ] Sending PUBLISH (d0, q0, r0, m187), 'b'raspberrypi/version'', ... (10 bytes)
2021-01-09 18:37:29,686 [TRACE] [mqtt    ] [mqtt.client             ] Received SUBACK
```

</details>

CrazyIvan359, 2021-01-09:
Doesn't look like a problem with MQTTany, seems like something else is using the pin.

If you run `gpioinfo` again, does this line show `[used]` at the end? (It did not in your original output you gave me the other day)
```
        line  26:     "GPIO26"       unused   input  active-high 
```

Veldhuis123, 2021-01-09:
i have run it again and he says unused

CrazyIvan359, 2021-01-09:
Could you try this test script. Paste it into a file (ex. `test.py`) and run it using `python3 test.py`.

<details><summary>test.py</summary>

```python
from periphery.gpio import GPIO, GPIOError

PIN_NUMBER = 26
PIN_MODE = "in"  # options are "in" and "out"
PIN_INTERRUPT = "both"  # options are "none", "rising", "falling", or "both"
PIN_RESISTOR = "pull_up"  # options are "default", "pull_up", "pull_down", or "disable"

if __name__ == "__main__":
    pin=None
    try:
        pin = GPIO(
            "/dev/gpiochip0",
            PIN_NUMBER,
            PIN_MODE,
            edge=PIN_INTERRUPT,
            bias=PIN_RESISTOR,
        )
    except GPIOError as err:
        print(err)
        exit(1)
    else:
        while True:
            if pin.poll(5):
                edge, ns = pin.read_event()
                print(
                    f"Interrupt fired on pin {PIN_NUMBER} at {ns}ns with {str(edge).upper()} edge"
                )
            else:
                print("No interrupt events in the last 5 seconds")
                print(
                    f"Pin {PIN_NUMBER} is currently {'HIGH' if pin.read() else 'LOW'}"
                )
    finally:
        if pin:
            pin.close()

```

</details

Veldhuis123, 2021-01-09:
when i will run this script 

```
[Errno 22] Opening input line event handle: Invalid argument

```
it could also be that I have configured something incorrectly in openhab?

CrazyIvan359, 2021-01-09:
No, openhab is not the problem, though looking closer you are trying to set the topic in MQTTany, but this will not work. Maybe look through the documentation again, if you were using a version prior to v0.12.0 then several things have changed since you last looked.

Maybe this is a permissions problem, though MQTTany should report that. What does this command show?

```bash
ls -lh /dev/gpiochip0
```

Veldhuis123, 2021-01-09:
this is the output of ``` ls -lh /dev/gpiochip0 ```
```
pi@raspberrypi:~ $ ls -lh /dev/gpiochip0
crw-rw---- 1 root gpio 254, 0 Dec 29 20:19 /dev/gpiochip0```

CrazyIvan359, 2021-01-09:
Ok, what about the command `groups`? Does that list your user as being in the `gpio` group?

Veldhuis123, 2021-01-09:
if I look at it that way, that's okay
```
pi@raspberrypi:~ $ groups pi
pi : pi adm dialout cdrom sudo audio video plugdev games users input netdev spi i2c gpio lpadmin
```

CrazyIvan359, 2021-01-09:
Ok, maybe sysfs has it exported. What does the output of `ls -lh /sys/class/gpio` show?

Veldhuis123, 2021-01-09:
this is the output of ```ls -lh /sys/class/gpio ```
 ```
pi@raspberrypi:~ $ ls -lh /sys/class/gpio
total 0
-rwxrwx--- 1 root gpio 4.0K Jan  9 19:45 export
lrwxrwxrwx 1 root gpio    0 Dec 29 20:19 gpiochip0 -> ../../devices/platform/soc/fe200000.gpio/gpio/gpiochip0
lrwxrwxrwx 1 root gpio    0 Dec 29 20:19 gpiochip500 -> ../../devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.0/gpio/gpiochip500
lrwxrwxrwx 1 root gpio    0 Dec 29 20:19 gpiochip504 -> ../../devices/platform/soc/soc:firmware/soc:firmware:gpio/gpio/gpiochip504
-rwxrwx--- 1 root gpio 4.0K Dec 29 20:19 unexport
 ```

CrazyIvan359, 2021-01-09:
Ok, nothing there. Modify these lines in the test script and run it again:

```python
PIN_INTERRUPT = "none"
PIN_RESISTOR = "default"
```

Veldhuis123, 2021-01-10:
i have run that script
```
pi@raspberrypi:~ $ python test.py
  File "test.py", line 28
    f"Interrupt fired on pin {PIN_NUMBER} at {ns}ns with {str(edge).upper()} edge"
                                                                                 ^
SyntaxError: invalid syntax
```

Veldhuis123, 2021-01-10:
Hello,

I think I managed to add mqtt to the gpio. but I just can't figure out how to operate gpio pin 26 with openhab 3.0. do you have any tips on how to get started?

```
 line  26:     "GPIO26"    "MQTTany"  output  active-high [used]
```

CrazyIvan359, 2021-01-10:
It does look like it is working. Did you change something to make it work?

---

This section on [Paths](https://crazyivan359.github.io/mqttany/start.html#paths) should help you understand how topics work, and each module also has a section detailing what nodes and properties it exposes. The one for the GPIO module can be found [here](https://crazyivan359.github.io/mqttany/interface/gpio/index.html#paths).

As an example, you would configure the pin we have been testing with like this:

```text
stateTopic="raspberrypi/gpio/channel1"
commandTopic="raspberrypi/gpio/channel1/set"
on="ON"
off="OFF"
```

The rest of your configuration looks good.

Veldhuis123, 2021-01-10:
i have change the configuration but it don't work.

at the moment I am not coping very well with it, but it is getting better
```
gpio:
  polling interval: 60
  debounce: 10

  channel1:
    pin: 26
    name: '{relay}'
    pin mode: 'output'
    resistor: 'off'
```

this is my openHab configuration

```
UID: mqtt:topic:openhab:relay
label: Generic MQTT Thing
thingTypeUID: mqtt:topic
configuration: {}
bridgeUID: mqtt:broker:openhab
channels:
  - id: channel1
    channelTypeUID: mqtt:switch
    label: test
    description: ""
    configuration:
      commandTopic: 10.0.10.200/relay/1/set
      stateTopic: 10.0.10.200/relay/1

```

CrazyIvan359, 2021-01-10:
Ok, please let me know if it keeps working now on your RPi4.

I will open a topic in the Discussions tab to discuss your openHAB configuration.

Veldhuis123, 2021-01-10:
okay is completely fine, i will see if it still works tomorrow.

CrazyIvan359, 2021-01-21:
@Veldhuis123 the issue you were seeing with the `error 22` is actually caused by the kernel not supporting setting the bias resistor. I have added a kernel version check so that MQTTany will still run without that capability. A warning will be written to the log to indicate that MQTTany was not able to set the bias resistor but it will work otherwise.

Since this seems to be the only issue anyone has had with an RPi4 I will be merging that code shortly so you will be able to use the master branch on your RPi4 without issue.

closed 2021-01-21 by CrazyIvan359

Veldhuis123, 2021-01-24:
thans for your help, but I can't get it to work

CrazyIvan359, 2021-01-25:
@Veldhuis123 I made a mistake in the update, and then another in the fix. I will get v0.14.2 published tonight that should fix it.

CrazyIvan359, 2021-01-28:
@Veldhuis123 the issue has been resolved and v0.14.2 is now available

PR #109 (2021-01-21): Add Raspberry Pi 4B support
Closes #79