CrazyIvan359/mqttany #80: Additional nested configuration option causes gpio module to not start anymore

sihui62, 2020-12-06:
When adding the configuration option `digital:` the gpio module does not start anymore.

### Configuration
```yaml
  ### Interrupt Debounce
  #debounce: 50 # in milliseconds

  ### Single Pin Configuration
  my-pin:

    ### Pin Number
    pin: 5

    ### Pin Name
    # Subsitutions:
    # - {pin} = pin number
    # - {pin:02d} = pin number padded with 0's to always be 2 numbers
    # - {pin_id} = name of this section
    #name: '{pin_id}'

    ### Pin Mode
    pin mode: 'output' # can be 'input' or 'output'
    digital:
    initial state: OFF

    ### Pin Resistor
    #resistor: 'off' # can be 'pullup', 'pulldown', or 'off'

    ### Pin Type Specifc Options
    # Each pin type will have some options specific to it,
    # consult the documentation for details.

  ### Multiple Pin Configuration
```

### Logs
```text
2020-12-06 08:20:50,306 [INFO ] [core                    ] MQTTany 0.12.0 starting
2020-12-06 08:20:50,550 [INFO ] [core.gpio               ] Detected board: RASPBERRY_PI_3B
2020-12-06 08:20:50,861 [ERROR] [gpio                    ] An exception occurred while running function 'load'
2020-12-06 08:20:50,863 [ERROR] [gpio                    ]

Traceback (most recent call last):
  File "/opt/mqttany/mqttany/common.py", line 319, in _call
    retval = func(**kwargs)
  File "/opt/mqttany/mqttany/modules/gpio/core.py", line 94, in load
    config_data = parse_config(config_raw, conf_options, log)
  File "/opt/mqttany/mqttany/config.py", line 256, in parse_config
    valid, config = parse_dict(data, options)
  File "/opt/mqttany/mqttany/config.py", line 245, in parse_dict
    config,
  File "/opt/mqttany/mqttany/config.py", line 139, in process_option
    section_valid, section_config = parse_dict(value, option)
  File "/opt/mqttany/mqttany/config.py", line 249, in parse_dict
    key, data.pop(key, "**NO DATA**"), options[key], config
  File "/opt/mqttany/mqttany/config.py", line 165, in process_option
    if not isinstance(value, option["type"]):
TypeError: isinstance() arg 2 must be a type or tuple of types

2020-12-06 08:20:50,864 [WARN ] [core                    ] Module 'gpio' load failed
2020-12-06 08:20:50,871 [INFO ] [core                    ] Module 'mqtt' started successfully
2020-12-06 08:20:50,897 [INFO ] [mqtt                    ] Connected to broker '192.168.2.211:1883'
```

After removing the `digital:` config option the module loads as expected:

```text
2020-12-06 08:23:53,857 [INFO ] [core                    ] MQTTany 0.12.0 starting
2020-12-06 08:23:54,100 [INFO ] [core.gpio               ] Detected board: RASPBERRY_PI_3B
2020-12-06 08:23:54,444 [INFO ] [core                    ] Module 'mqtt' started successfully
2020-12-06 08:23:54,452 [INFO ] [core                    ] Module 'gpio' started successfully
2020-12-06 08:23:54,462 [INFO ] [gpio.digital            ] Setting up 'my-pin' on GPIO05 as OUTPUT
2020-12-06 08:23:54,477 [INFO ] [mqtt                    ] Connected to broker '192.168.2.211:1883'
```

CrazyIvan359, 2020-12-06:
So first thing I'm seeing is that you don't have correct indentation. I'm spinning up a new SD card for an RPi0 to test, maybe you'll be faster. Try the following for your configuration, it may not work though since the error you are showing me here is different than the one you showed me on the forum.

```yaml
  my-pin:

    ### Pin Number
    pin: 5

    ### Pin Name
    # Subsitutions:
    # - {pin} = pin number
    # - {pin:02d} = pin number padded with 0's to always be 2 numbers
    # - {pin_id} = name of this section
    #name: '{pin_id}'

    ### Pin Mode
    pin mode: 'output' # can be 'input' or 'output'
    digital:
      initial state: OFF
```

sihui62, 2020-12-06:
I won't be able to do that before tomorrow as I am at work today, sorry.

CrazyIvan359, 2020-12-06:
So with the valid indentation (which I notice you did have in the forum, with the other error) I get the following error in the log:

```log
2020-12-06 17:27:15,621 [ERROR] [core    ] [gpio                    ] Value 'False' for config option 'initial state' is not one of ['ON', 'on', 'OFF', 'off']
2020-12-06 17:27:15,622 [WARN ] [core    ] [gpio                    ] Optional section 'digital' is not valid, it will be ignored
```

which explains the error you see farther on in the log saying that the `digital` key is missing:

```log
2020-12-06 17:27:15,668 [ERROR] [core    ] [gpio                    ] An exception occurred while running function 'load'
2020-12-06 17:27:15,672 [ERROR] [core    ] [gpio                    ] 

Traceback (most recent call last):
  File "/opt/mqttany/mqttany/common.py", line 319, in _call
    retval = func(**kwargs)
  File "/opt/mqttany/mqttany/modules/gpio/core.py", line 116, in load
    pin_object = build_pin(id, named_config)
  File "/opt/mqttany/mqttany/modules/gpio/core.py", line 83, in build_pin
    clazz = clazz(pin, CONFIG[CONF_KEY_MODE], id, name, pin_config)
  File "/opt/mqttany/mqttany/modules/gpio/pin/digital.py", line 119, in __init__
    self._edge = pin_config[CONF_KEY_DIGITAL][CONF_KEY_INTERRUPT]
KeyError: 'digital'

```

The config parser incorrectly reports that the configuration is valid, even though the `digital` section has an invalid value. This partly makes sense, because the `digital` section is a conditional section and the code handling that is related to the optional section code. I need to change the handling of invalid values in sections with conditions that are true.

I will get a patch done up shortly to fix this.

CrazyIvan359, 2020-12-06:
Problem 2 is a YAML one... The values `on` and `off` (case in-sensitive) are evaluated to `True` and `False`, respectively, by the YAML parser! MQTTany's config parser is expect a string not a boolean.

PR #82 (2020-12-06): Fix yaml on/off boolean handling
The YAML parser used for reading the configuration considers `on` or `off` (case in-sensitive) to be boolean values, this leads to confusion when using them without quotes when a string is expected. 

Boolean options have been added to selection lists where applicable and the config parser no longer converts the value to a string before attempt to match a selection key.

Fixes #80
closed 2020-12-06 by CrazyIvan359