CrazyIvan359/mqttany #102: Error when start MQTTany

Veldhuis123, 2020-12-30:
when i start MQTTany get i this fault

2020-12-30 14:27:33,770 [ERROR] [core                    ] 

Traceback (most recent call last):
  File "mqttany/mqttany.py", line 110, in <module>
    core.start(queue_core, args.config_file)
  File "/opt/mqttany/mqttany/core.py", line 382, in start
    import gpio
  File "/opt/mqttany/mqttany/gpio/__init__.py", line 47, in <module>
    board = get_board()
  File "/opt/mqttany/mqttany/gpio/boards/__init__.py", line 56, in get_board
    return all_boards.get(Detector().board.id, Unknown)()
  File "/opt/mqttany/mqttany/gpio/boards/__init__.py", line 42, in __init__
    self._chips.append(999)  # dummy
AttributeError: 'Unknown' object has no attribute '_chips'

2020-12-30 14:27:33,770 [WARN ] [core                    ] MQTTany exiting with errors

CrazyIvan359, 2020-12-30:
A bug indeed, I missed these `Board` subclasses when I changed from `__init_subclass__`  to `__init__` in #96. I have made #103 to fix it.

The fact that it attempts to use the `Unknown` class means that the board you are on is not supported. If you want me to add support for it please open a board request issue, the template contains some commands to run that will give me the information I need.

Please use code fences when posting code or logs like so:

````
```log

```
````

closed 2020-12-30 by CrazyIvan359

Veldhuis123, 2021-01-02:
Hello,

Thanks for your reaction

Robin Veldhuis

> Op 30 dec. 2020, om 21:04 heeft Michael Murton <notifications@github.com> het volgende geschreven:
> 
> Closed #102 <https://github.com/CrazyIvan359/mqttany/issues/102> via #103 <https://github.com/CrazyIvan359/mqttany/pull/103>.

PR #103 (2020-12-30): Fix Generic + Unknown board classes
Fex *Unknown* and *Generic Linux* boards missing superclass init.

This bug was introduced in #96 when the base board class changed from using `__init_subclass__` to `__init__`.

Fixes #102