CrazyIvan359/mqttany #120: Bug Report

EKNYESE, 2021-08-01:
### Describe the bug
"pip3 install -r requirements/onewire.txt" not possible since missing!!

### Expected behavior
"pip3 install -r requirements/onewire.txt" possible

### Environment
- MQTTany version: actual one
- Platform: openHABian on RPi4

### Configuration
```yaml
# n/a because of missing example (although I have invested severals hours it was still not possible for me to get en example how to patch the config for several DS18B20..
```

### Logs
```text
Insert relevant log entries within code fences here,
or attach the entire log as a file.
```

CrazyIvan359, 2021-08-01:
Looks like I did not remove the note in the docs to install One Wire requirements, there are none anymore. I wrote my own One Wire interface library so no external ones are needed.

What else are you hung up on getting your sensors working?

EKNYESE, 2021-08-02:
Unfortunately I was not able until now to use it for DS18B20. Maybe it is a problem with user-rights - I am using openhabian as base.

Do You have an idea, how to interpret this failure message?:
Could you please also give a concrete example, how to patch the config to read DS18B20? Right now I am not understanding the documentation and I am happy, that I have set it up right for reading GPIOs (also after several hours of tries because of spaces - needed by python etc..).. 

[12:42:25] openhabian@openHABianRPi4:~$ /usr/bin/python3 /opt/mqttany/mqttany/mqttany.py
2021-08-02 12:42:37,987 [INFO ] [core                    ] MQTTany 0.14.3 starting
2021-08-02 12:42:38,074 [INFO ] [core.gpio               ] Detected board: RASPBERRY_PI_4B
2021-08-02 12:42:38,309 [ERROR] [onewire                 ] An exception occurred while running function 'lo                                     ad'
2021-08-02 12:42:38,311 [ERROR] [onewire                 ]

Traceback (most recent call last):
  File "/opt/mqttany/mqttany/modules/__init__.py", line 83, in call
    retval = func(**kwargs)
  File "/opt/mqttany/mqttany/modules/onewire/core.py", line 129, in load
    config_data = parse_config(config_raw, conf_options, log)
  File "/opt/mqttany/mqttany/config.py", line 261, in parse_config
    valid, config = parse_dict(data, options)
  File "/opt/mqttany/mqttany/config.py", line 250, in parse_dict
    config,
  File "/opt/mqttany/mqttany/config.py", line 170, in process_option
    if not isinstance(value, option["type"]):
TypeError: isinstance() arg 2 must be a type or tuple of types

2021-08-02 12:42:38,312 [WARN ] [core                    ] Module 'onewire' load failed

CrazyIvan359, 2021-08-02:
That is possibly an issue with the One Wire module, I will need some more information to dig into it though. Please run MQTTany with the `-vv` option and paste the log from the start up to the error here. I would also like to see your One Wire section from the config file. 

Please use code fences when posting logs or configs so that they are readable. Here are examples:

````
```log
Paste log here
```

```yaml
Paste config here
```
````

As for the possible access issues, I don't remember if openhabian has the udev rules or not, but you should follow the instructions on the One Wire page to install the ones included with MQTTany.

CrazyIvan359, 2021-08-03:
I edited your comment to use collapsible sections because I'm on my phone.

So the udev rules file is [here](https://github.com/CrazyIvan359/mqttany/blob/master/udev/98-w1.rules) which if you followed the instructions to the letter should be in `/opt/mqttany/udev`.

There is definitely a bug in the One Wire module though, your config is fine. I will look into that later this week when I have time.

As for getting One Wire configured and also your GPIO working, because the log is telling me it isn't, let's switch to a discussion which I will link here in a minute when I create it.

CrazyIvan359, 2021-12-26:
@EKNYESE sorry it took so long to get back to this, I have not had much time to work on MQTTany this year.

I came to work on this issue but looking through everything here again I found that the problem is actually with your config. This key in the `onewire` section is wrong:

```yaml
  ### Bus Interface
  bus: 'w1'
```

It should be 'bus interface' and not 'bus'

```yaml
  ### Bus Interface
  bus interface: 'w1'
```

closed 2021-12-26 by CrazyIvan359