CrazyIvan359/mqttany #127: Polling resets states

Elch2000, 2021-11-07:
### Describe the bug
Polling GPIO states resets the state to off of output pins

### Expected behavior
Polling reads the current state and does not modify the state

### Environment
- MQTTany version: 0.14.3
- Modules: MQTT, GPIO
- Hardware: RASPBERRY_PI_4B

### Logs
```text
2021-11-07 21:29:03,829 [DEBUG] [gpio    ] [gpio                    ] Polling timer fired
2021-11-07 21:29:03,831 [DEBUG] [gpio    ] [gpio                    ] Polling all pins
2021-11-07 21:29:03,833 [DEBUG] [gpio    ] [gpio.digital            ] Read state OFF logic LOW from 'line 0' on GPIO26
2021-11-07 21:29:03,835 [DEBUG] [gpio    ] [gpio.digital            ] Read state ON logic HIGH from 'line 1' on GPIO19
2021-11-07 21:29:03,835 [TRACE] [core    ] [bus.transmit            ] Message received on publish queue: PublishMessage('gpio/line-0', 'OFF')
2021-11-07 21:29:03,837 [DEBUG] [gpio    ] [gpio.digital            ] Read state ON logic HIGH from 'line 2' on GPIO13
2021-11-07 21:29:03,838 [TRACE] [core    ] [bus.transmit            ] Message received on publish queue: PublishMessage('gpio/line-1', 'ON')
2021-11-07 21:29:03,839 [DEBUG] [gpio    ] [gpio.digital            ] Read state ON logic HIGH from 'line 3' on GPIO06
2021-11-07 21:29:03,841 [TRACE] [core    ] [bus.transmit            ] Message received on publish queue: PublishMessage('gpio/line-2', 'ON')
2021-11-07 21:29:03,842 [DEBUG] [gpio    ] [gpio.digital            ] Read state ON logic HIGH from 'line 4' on GPIO05
2021-11-07 21:29:03,842 [TRACE] [mqtt    ] [mqtt                    ] Message received to transmit: PublishMessage('gpio/line-0', 'OFF')
2021-11-07 21:29:03,844 [TRACE] [core    ] [bus.transmit            ] Message received on publish queue: PublishMessage('gpio/line-3', 'ON')
2021-11-07 21:29:03,845 [DEBUG] [gpio    ] [gpio.digital            ] Read state ON logic HIGH from 'line 5' on GPIO11
2021-11-07 21:29:03,846 [TRACE] [mqtt    ] [mqtt.client             ] Sending PUBLISH (d0, q0, r0, m21), 'b'raspberrypi/gpio/line-0'', ... (3 bytes)
2021-11-07 21:29:03,847 [DEBUG] [gpio    ] [gpio.digital            ] Read state ON logic HIGH from 'line 6' on GPIO09
2021-11-07 21:29:03,849 [DEBUG] [gpio    ] [gpio.digital            ] Read state ON logic HIGH from 'line 7' on GPIO10
2021-11-07 21:29:03,847 [TRACE] [core    ] [bus.transmit            ] Message received on publish queue: PublishMessage('gpio/line-4', 'ON')
2021-11-07 21:29:03,850 [TRACE] [mqtt    ] [mqtt                    ] Message received to transmit: PublishMessage('gpio/line-1', 'ON')
2021-11-07 21:29:03,851 [DEBUG] [gpio    ] [gpio.digital            ] Read state ON logic HIGH from 'line 8' on GPIO22
2021-11-07 21:29:03,853 [TRACE] [mqtt    ] [mqtt.client             ] Sending PUBLISH (d0, q0, r0, m22), 'b'raspberrypi/gpio/line-1'', ... (2 bytes)
2021-11-07 21:29:03,854 [DEBUG] [gpio    ] [gpio.digital            ] Read state ON logic HIGH from 'line 9' on GPIO27
2021-11-07 21:29:03,853 [TRACE] [core    ] [bus.transmit            ] Message received on publish queue: PublishMessage('gpio/line-5', 'ON')
2021-11-07 21:29:03,856 [DEBUG] [gpio    ] [gpio.digital            ] Read state ON logic HIGH from 'line 10' on GPIO17
2021-11-07 21:29:03,858 [TRACE] [mqtt    ] [mqtt                    ] Message received to transmit: PublishMessage('gpio/line-2', 'ON')
2021-11-07 21:29:03,857 [TRACE] [core    ] [bus.transmit            ] Message received on publish queue: PublishMessage('gpio/line-6', 'ON')
2021-11-07 21:29:03,858 [DEBUG] [gpio    ] [gpio.digital            ] Read state ON logic HIGH from 'line 11' on GPIO04
2021-11-07 21:29:03,860 [DEBUG] [gpio    ] [gpio.digital            ] Read state OFF logic LOW from 'led 0' on GPIO12
2021-11-07 21:29:03,861 [TRACE] [mqtt    ] [mqtt.client             ] Sending PUBLISH (d0, q0, r0, m23), 'b'raspberrypi/gpio/line-2'', ... (2 bytes)
2021-11-07 21:29:03,860 [TRACE] [core    ] [bus.transmit            ] Message received on publish queue: PublishMessage('gpio/line-7', 'ON')
2021-11-07 21:29:03,862 [DEBUG] [gpio    ] [gpio.digital            ] Read state OFF logic LOW from 'led 1' on GPIO16
2021-11-07 21:29:03,863 [TRACE] [mqtt    ] [mqtt                    ] Message received to transmit: PublishMessage('gpio/line-3', 'ON')
2021-11-07 21:29:03,864 [TRACE] [core    ] [bus.transmit            ] Message received on publish queue: PublishMessage('gpio/line-8', 'ON')
2021-11-07 21:29:03,864 [DEBUG] [gpio    ] [gpio.digital            ] Read state OFF logic LOW from 'led 2' on GPIO20
2021-11-07 21:29:03,866 [TRACE] [mqtt    ] [mqtt.client             ] Sending PUBLISH (d0, q0, r0, m24), 'b'raspberrypi/gpio/line-3'', ... (2 bytes)
2021-11-07 21:29:03,866 [DEBUG] [gpio    ] [gpio.digital            ] Read state OFF logic LOW from 'led 3' on GPIO21
2021-11-07 21:29:03,867 [TRACE] [core    ] [bus.transmit            ] Message received on publish queue: PublishMessage('gpio/line-9', 'ON')
2021-11-07 21:29:03,869 [TRACE] [mqtt    ] [mqtt                    ] Message received to transmit: PublishMessage('gpio/line-4', 'ON')
2021-11-07 21:29:03,870 [TRACE] [core    ] [bus.transmit            ] Message received on publish queue: PublishMessage('gpio/line-10', 'ON')
2021-11-07 21:29:03,871 [TRACE] [mqtt    ] [mqtt.client             ] Sending PUBLISH (d0, q0, r0, m25), 'b'raspberrypi/gpio/line-4'', ... (2 bytes)
2021-11-07 21:29:03,872 [TRACE] [core    ] [bus.transmit            ] Message received on publish queue: PublishMessage('gpio/line-11', 'ON')
2021-11-07 21:29:03,874 [TRACE] [core    ] [bus.transmit            ] Message received on publish queue: PublishMessage('gpio/led-0', 'OFF')
2021-11-07 21:29:03,874 [TRACE] [mqtt    ] [mqtt                    ] Message received to transmit: PublishMessage('gpio/line-5', 'ON')
2021-11-07 21:29:03,876 [TRACE] [core    ] [bus.transmit            ] Message received on publish queue: PublishMessage('gpio/led-1', 'OFF')
2021-11-07 21:29:03,878 [TRACE] [mqtt    ] [mqtt.client             ] Sending PUBLISH (d0, q0, r0, m26), 'b'raspberrypi/gpio/line-5'', ... (2 bytes)
2021-11-07 21:29:03,878 [TRACE] [core    ] [bus.transmit            ] Message received on publish queue: PublishMessage('gpio/led-2', 'OFF')
2021-11-07 21:29:03,880 [TRACE] [core    ] [bus.transmit            ] Message received on publish queue: PublishMessage('gpio/led-3', 'OFF')
2021-11-07 21:29:03,881 [TRACE] [mqtt    ] [mqtt                    ] Message received to transmit: PublishMessage('gpio/line-6', 'ON')
2021-11-07 21:29:03,883 [TRACE] [mqtt    ] [mqtt.client             ] Sending PUBLISH (d0, q0, r0, m27), 'b'raspberrypi/gpio/line-6'', ... (2 bytes)
2021-11-07 21:29:03,885 [TRACE] [mqtt    ] [mqtt                    ] Message received to transmit: PublishMessage('gpio/line-7', 'ON')
2021-11-07 21:29:03,887 [TRACE] [mqtt    ] [mqtt.client             ] Sending PUBLISH (d0, q0, r0, m28), 'b'raspberrypi/gpio/line-7'', ... (2 bytes)
2021-11-07 21:29:03,889 [TRACE] [mqtt    ] [mqtt                    ] Message received to transmit: PublishMessage('gpio/line-8', 'ON')
2021-11-07 21:29:03,890 [TRACE] [mqtt    ] [mqtt.client             ] Sending PUBLISH (d0, q0, r0, m29), 'b'raspberrypi/gpio/line-8'', ... (2 bytes)
2021-11-07 21:29:03,892 [TRACE] [mqtt    ] [mqtt                    ] Message received to transmit: PublishMessage('gpio/line-9', 'ON')
2021-11-07 21:29:03,894 [TRACE] [mqtt    ] [mqtt.client             ] Sending PUBLISH (d0, q0, r0, m30), 'b'raspberrypi/gpio/line-9'', ... (2 bytes)
2021-11-07 21:29:03,900 [TRACE] [mqtt    ] [mqtt                    ] Message received to transmit: PublishMessage('gpio/line-10', 'ON')
2021-11-07 21:29:03,903 [TRACE] [mqtt    ] [mqtt.client             ] Sending PUBLISH (d0, q0, r0, m31), 'b'raspberrypi/gpio/line-10'', ... (2 bytes)
2021-11-07 21:29:03,907 [TRACE] [mqtt    ] [mqtt                    ] Message received to transmit: PublishMessage('gpio/line-11', 'ON')
2021-11-07 21:29:03,909 [TRACE] [mqtt    ] [mqtt.client             ] Sending PUBLISH (d0, q0, r0, m32), 'b'raspberrypi/gpio/line-11'', ... (2 bytes)
2021-11-07 21:29:03,912 [TRACE] [mqtt    ] [mqtt                    ] Message received to transmit: PublishMessage('gpio/led-0', 'OFF')
2021-11-07 21:29:03,913 [TRACE] [mqtt    ] [mqtt.client             ] Sending PUBLISH (d0, q0, r0, m33), 'b'raspberrypi/gpio/led-0'', ... (3 bytes)
2021-11-07 21:29:03,914 [TRACE] [mqtt    ] [mqtt                    ] Message received to transmit: PublishMessage('gpio/led-1', 'OFF')
[... cut ...]
2021-11-07 21:29:03,921 [TRACE] [mqtt    ] [mqtt.client             ] Sending PUBLISH (d0, q0, r0, m36), 'b'raspberrypi/gpio/led-3'', ... (3 bytes)
2021-11-07 21:29:04,925 [TRACE] [mqtt    ] [mqtt.client             ] Sending PINGREQ
2021-11-07 21:29:04,927 [TRACE] [mqtt    ] [mqtt.client             ] Received PINGRESP

```

"led-2" was ON but after polling cycle it turned OFF. This behavior has been reproduced on every output port (led-0 .. led-3). When the polling interval is set to 0 led-2 stays turned on.

Please let me know if more information is required.

Cheers

closed 2021-11-07 by Elch2000

CrazyIvan359, 2021-11-08:
Hey @Elch2000 did you close this because you found the issue? What was the problem in the end?

Elch2000, 2021-11-08:
Hi @CrazyIvan359,

I have a completely other problem: Sending an ON command to led-2 does not turn on the LED. So MQTTany was completely right when polling the state... I did not find the problem yet. And currently no idea how to fix this :(

Cheers

Elch2000, 2021-11-10:
Hi @CrazyIvan359,

it looks like the MQTTany instance does not recognize MQTT commands in spite of being registered to the broker correctly. 

To rule out a hardware or OS related problem I installed another application (MQTT IO, https://mqtt-io.app) on the same hardware (Pi4B). This application toggles a LED flawlessly when it receives a corespondending MQTT command.

What do you suggest for fixing this problem? Reopen this issue? Or anything else?

Cheers

CrazyIvan359, 2021-11-11:
Without more details my guess would be wrong MQTT topic. Looking at your log it should be `raspberrypi/gpio/led-2/set` to command `led-2`.

Elch2000, 2021-11-12:
Hi,

you are right! I forgot to append "/set". AAARGH. Sorry for disturbing you...

 Thank you for pointing to that.

Cheers