CrazyIvan359/mqttany #142: MQTTany 0.14.3 exception running against Python 3.10.4

guitarglen, 2022-09-14:
Hi there,
I have had to rebuild the PI controller I am running MQTTany on, and am now getting an exception. 

It appears to be because I am rebuilding/running the latest version of Ubuntu and therefore Python is a newer version... From what I can see, this issue stems from the newer version of python, and possibly would be resolved from downgrading to 3.9. But I am wondering if there is any possibility of updating this MQTTany dependency to work with Python 3.10 as well?

Details below. 

Cheers - Glen 

Versions:
- MQTTany: 0.14.3
- Python: 3.10.4
- Enironment: Rasberry PI 3 /Ubuntu 22.04.1 LTS

`2022-08-10 00:36:34,859 [INFO ] [core                    ] MQTTany 0.14.3 starting
2022-08-10 00:36:34,987 [INFO ] [core.gpio               ] Detected board: RASPBERRY_PI_3B
2022-08-10 00:36:35,297 [ERROR] [gpio                    ] An exception occurred while running function 'load'
2022-08-10 00:36:35,301 [ERROR] [gpio                    ] 

Traceback (most recent call last):
  File "/opt/mqttany/mqttany/modules/__init__.py", line 83, in call
    retval = func(**kwargs)
  File "/opt/mqttany/mqttany/modules/gpio/core.py", line 98, in load
    conf_options = updateConfOptions(CONF_OPTIONS)
  File "/opt/mqttany/mqttany/modules/gpio/pin/__init__.py", line 57, in updateConfOptions
    conf_options = update_dict(conf_options, digital.CONF_OPTIONS)
  File "/opt/mqttany/mqttany/common.py", line 365, in update_dict
    if not isinstance(dv, collections.MutableMapping):  # type: ignore
AttributeError: module 'collections' has no attribute 'MutableMapping'

2022-08-10 00:36:35,302 [WARN ] [core                    ] Module 'gpio' load failed
`

CrazyIvan359, 2022-09-14:
Your diagnosis is correct, though I should have switched to using `collections.abc` long ago. I'll make a patch now since that's an easy fix and I'll get you to test it since I don't actually have a board handy that's running it due to renovations.

Also, for inline code you use single tildes `` `like this` `` but for multi-line blocks you need to use at least 3 before and after like this
````
```
code
block
```
````

CrazyIvan359, 2022-09-14:
@guitarglen all set, there is a branch `python-updates` that should fix your issue. I ended up bumping the minimum Python version to 3.7 at the same time as there were several other issues happening when running the ci tests. I ended up bumping the platform detect version at the same time, so if you want to try that out for me as well use the branch `update-platform-detect` instead (it includes the above fixes).

Let me know if these are working for you!

guitarglen, 2022-09-15:
Next test is for the update-platform-detect branch: 
`sudo git clone -b update-platform-detect https://github.com/CrazyIvan359/mqttany.git`

Again starts up cleanly and shuts down cleanly

As an aside, I think **paho-mqtt** is missing from the PIP requirements file for all branches, including master.... Not a biggie, but if you are dropping a new release, may pay to check that too.
```
/opt/mqttany$ python3 --version
Python 3.10.4

/opt/mqttany$ sudo python3 mqttany/mqttany.py
2022-09-15 20:25:48,850 [INFO ] [core                    ] MQTTany 0.14.3-dev starting
2022-09-15 20:25:48,851 [WARN ] [core                    ] 
2022-09-15 20:25:48,852 [WARN ] [core                    ] ########  DEVELOPMENT VERSION  ########
2022-09-15 20:25:48,854 [WARN ] [core                    ] #                                     #
2022-09-15 20:25:48,855 [WARN ] [core                    ] #  This is a development version and  #
2022-09-15 20:25:48,856 [WARN ] [core                    ] #    may be incomplete or unstable    #
2022-09-15 20:25:48,857 [WARN ] [core                    ] #                                     #
2022-09-15 20:25:48,858 [WARN ] [core                    ] #######################################
2022-09-15 20:25:48,860 [WARN ] [core                    ] 
2022-09-15 20:25:49,076 [INFO ] [core.gpio               ] Detected board: RASPBERRY_PI_3B
2022-09-15 20:25:49,506 [INFO ] [core                    ] Module 'mqtt' started successfully
2022-09-15 20:25:49,519 [INFO ] [core                    ] Module 'gpio' started successfully
2022-09-15 20:25:49,533 [INFO ] [gpio.digital            ] Setting up 'gatebuttonoutside' on GPIO05 as INPUT
2022-09-15 20:25:49,542 [INFO ] [gpio.digital            ] Setting up 'gatebuttoninside' on GPIO06 as INPUT
2022-09-15 20:25:49,546 [INFO ] [gpio.digital            ] Setting up 'gateremoteopen' on GPIO12 as INPUT
2022-09-15 20:25:49,550 [INFO ] [gpio.digital            ] Setting up 'gateopentrigger' on GPIO23 as OUTPUT
2022-09-15 20:25:49,553 [INFO ] [gpio.digital            ] Setting up 'gateclosetrigger' on GPIO24 as OUTPUT
2022-09-15 20:25:49,555 [INFO ] [gpio.digital            ] Setting up 'gatebuttonlights' on GPIO25 as OUTPUT
2022-09-15 20:25:49,558 [INFO ] [gpio.digital            ] Setting up 'gategardenlights' on GPIO26 as OUTPUT
2022-09-15 20:25:49,560 [INFO ] [gpio.digital            ] Setting up 'gateinsidebuttondisable' on GPIO27 as OUTPUT
2022-09-15 20:25:49,562 [INFO ] [gpio.digital            ] Setting up 'gateoutsidebuttondisable' on GPIO17 as OUTPUT
2022-09-15 20:25:49,564 [INFO ] [gpio.digital            ] Setting up 'gateirilluminatordisable' on GPIO04 as OUTPUT
2022-09-15 20:25:49,660 [INFO ] [mqtt                    ] Connected to broker 'remsrv:1883'
^C
2022-09-15 20:26:40,622 [INFO ] [core                    ] Module 'gpio' unloaded
2022-09-15 20:26:40,643 [INFO ] [core                    ] Module 'mqtt' unloaded
2022-09-15 20:26:41,076 [INFO ] [core                    ] MQTTany stopped
```
I have tested a few GPIO triggers etc, and work nicely as well. 

I will pop this version into the 'prod' controller, set it up to run as a service,  and let this run for while - Gets used reasonably frequently, so should get some good indication of any issues.
Thanks again.

CrazyIvan359, 2022-09-15:
Awesome, glad it's working for you and thanks for testing! I'll get those changes merged into master.

I actually moved the requirement into a file for MQTT communications. When I did the big redesign of the core I also turned the MQTT module into a pluggable one, thus leaving the door open for different communication methods. Following the pluggable pattern I moved the requirements into their own file.

https://github.com/CrazyIvan359/mqttany/blob/bef6ee60db50ccef3025af56ac165d21ea79cb18/requirements/mqtt.txt#L1

PR #144 (2022-09-15): Bump minimum Python version to 3.7
Closes #142
closed 2022-09-15 by CrazyIvan359