CrazyIvan359/mqttany #152: Interrupts on inputs do not work

zolee28, 2023-01-20:
### Describe the bug
I'm trying to use mqttany with Openhab. Connection over MQTT is good, I can send commands to the I2C and control my lights. However I also want to use MCP23017 for inputs (button press). The inputs work as well, the problem is that the polling interval of 1sec is not optimal for me.

I saw that you have been working on interrupts for the pins, now this could really help as when you press the button it would immediately sending its status to mqtt/openhab.

For me the interrupts do not work. I can show the pin changing in the regular poll intervals.

Is there something I did wrong in the config file? I also saw this does not have any effect on polling "interrupt polling interval: 50"

### Environment
- MQTTany version: v.0.14.3
- Platform: Raspberry Pi 4

### Configuration
```yaml
i2c:

  polling interval: 20

  mcp23017-1:

    interrupt polling interval: 50
    device: 'mcp23017'
    bus: 1
    address: 0x20
    mcp230xx:
      pin-a0:
        pin: 9
        direction: 'output'
        resistor: 'pullup'

  mcp23017-2:

    interrupt polling interval: 50
    device: 'mcp23017'
    bus: 1
    address: 0x21
    mcp230xx:
      pin-a0:
        pin: 5
        direction: 'input'
        resistor: 'pullup'
        interrupt: 'both'

```

### Logs
Nothing comes up on debug logs regarding interrupts.

CrazyIvan359, 2023-01-21:
The polling interval option should be under `mcp230xx`, it's not picking up the option where it is. 

Docs for that feature aren't merged, but you can find the updated MCP page [here](https://github.com/CrazyIvan359/mqttany/blob/docs-mcp230xx-interrupts/source/interface/i2c/devices/mcp230xx.rst)

closed 2023-01-21 by CrazyIvan359

zolee28, 2023-01-21:
Thanks a lot, works like a charm now!

So now the code looks like this;
```yaml

     mcp230xx:
       interrupt polling interval: 50
       pin-a4:
         pin: 4
         direction: 'input'
         resistor: 'pullup'
         interrupt: 'both'

       pin-a5:
         pin: 5
         direction: 'input'
         resistor: 'pullup'
         interrupt: 'both'

```
I ran into a new issue however, so for ex. when pin-a5 is switched, it sends the command OFF - this is ok. When I press pin-a4, pin a5 goes to ON and pin a4 is OFF. Is there a way that the status of the pins do not affect one another?